l5yth/potato-mesh #144: Docker tests fail

l5yth, 2025-09-21:
https://github.com/l5yth/potato-mesh/actions/runs/17890966688/job/50871343595

deuxyeux, 2025-09-21:
potatomesh-web container fails with

```
<internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require': cannot load such file -- sqlite3/sqlite3_native (LoadError)
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require'
        from /usr/local/bundle/gems/sqlite3-1.7.3-x86_64-linux/lib/sqlite3.rb:6:in `rescue in <top (required)>'
        from /usr/local/bundle/gems/sqlite3-1.7.3-x86_64-linux/lib/sqlite3.rb:2:in `<top (required)>'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:141:in `require'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:141:in `rescue in require'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:135:in `require'
        from app.rb:23:in `<main>'
<internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require': Error relocating /usr/local/bundle/gems/sqlite3-1.7.3-x86_64-linux/lib/sqlite3/3.3/sqlite3_native.so: posix_fallocate64: symbol not found - /usr/local/bundle/gems/sqlite3-1.7.3-x86_64-linux/lib/sqlite3/3.3/sqlite3_native.so (LoadError)
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:136:in `require'
        from /usr/local/bundle/gems/sqlite3-1.7.3-x86_64-linux/lib/sqlite3.rb:4:in `<top (required)>'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:141:in `require'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:141:in `rescue in require'
        from <internal:/usr/local/lib/ruby/3.3.0/rubygems/core_ext/kernel_require.rb>:135:in `require'
        from app.rb:23:in `<main>'
```

l5yth, 2025-09-22:
@deuxyeux try again please. should be fixed now.