l5yth/potato-mesh #194: Feature: Enable/Disable chat environment option

dynomite567, 2025-09-28:
It would be nice if administrators had the option to enable or disable the chat on a PotatoMesh instance, as some people might prefer to only show a map to the public and keep the chat to the mesh for privacy reasons.

I’ve included an example on how this would look on the website based on my own modifications on the code myself.

[image]

l5yth, 2025-10-03:
Can you try? #204

Run the app with `PRIVATE=1`

dynomite567, 2025-10-04:
Ran the branch in a separate directory and it looks like it works. Chat is hidden and everything else seems to function.

Here's the command I ran: ``SITE_NAME="Las Vegas Meshtastic Map" DEFAULT_CHANNEL="#LongFast" DEFAULT_FREQUENCY="915MHz" MAX_NODE_DISTANCE_KM=35 MATRIX_ROOM="" API_TOKEN="<api-token>" PRIVATE=1 APP_ENV=production bash app.sh``

[image]

PR #204 (2025-10-04): Add private mode to hide chat and message APIs
## Summary
- add a PRIVATE mode flag that removes chat UI content, disables the message API endpoints, and updates metadata copy
- filter hidden client nodes from API responses while wiring the private-mode flag to the frontend so chat fetches are skipped
- document the new PRIVATE environment variable and cover the behaviour with new RSpec examples
- fix #194
closed 2025-10-04 by l5yth